tbird20d/grabserial #8: Using -c <command> or --command="<command>"

wojt, 2015-03-25:
Hi,

I noticed that for command to work properly I had to change the code

from:
```Python
if command:
	sd.write(command + "\n")
	sd.flush()
```
to:
```Python
if command:
	sd.write(command + "\r")
	sd.flush()
```

Thanks,
Wojtek

tbird20d, 2015-03-27:
Are you, by chance, using this on Windows?

wojt, 2015-03-27:
Hi,

Nope, I'm using it on Ubuntu Linux. I'm working with Intel Galileo 2 board. Grabserial uses command syntax to interface with UEFI shell (EDK II).

Wojtek

wojt, 2015-03-27:
This may be related to UEFI shell. Its syntax is a little bit Window-ish. I'm not sure.

wojt, 2015-03-27:
Yup, sorry for spamming you with false issues.

I found this:
"Also note that UEFI uses CRLF line terminators (\r\n) instead of just LF (\n), and when we use native EFI functions, there is no layer which reinterprets LF to CRLF. Normally, applications use additional library called EFILIB which does the LF->CRLF transform."

Here:
http://x86asm.net/articles/uefi-programming-first-steps/

So it's just UEFI's shell issue. So unless you add a switch for grabserial to pick what line endings are used, then nothing can be done about it.

tbird20d, 2015-03-31:
>________________________________
> From: wojt <notifications@github.com>
>To: tbird20d/grabserial <grabserial@noreply.github.com> 
>Cc: Tim Bird <tbird20d@yahoo.com> 
>Sent: Friday, March 27, 2015 8:44 AM
>Subject: Re: [grabserial] Using -c <command> or --command="<command>" (#8)
> 
>
>
>Yup, sorry for spamming you with false issues.
>I found this:
>"Also note that UEFI uses CRLF line terminators (\r\n) instead of just LF (\n), and when we use native EFI functions, there is no layer which reinterprets LF to CRLF. Normally, applications use additional library called EFILIB which does the LF->CRLF transform."
>Here:
>http://x86asm.net/articles/uefi-programming-first-steps/
>So it's just UEFI's shell issue. So unless you add a switch for grabserial to pick what line endings are used, then nothing can be done about it.

I thought about this over the weekend, and I had two ideas:
1) just use \r\n always, and see if it's an issue for other shells or platforms.
2) try to automatically detect if we're in a UEFI shell, or alternatively in a shell that needs a '\r'.
I might be able to run a little test program to detect that.

I would like to avoid adding a switch if possible.

Let me know what you think.
 -- Tim

closed 2016-05-10 by tbird20d