tbird20d/grabserial #10: /dev/ttyAMA0 not found when using command line, but when edited in grabserial.py, works fine

kd5byb, 2015-08-23:
Good evening!

Was happy to find grabserial - I want to use a Raspberry Pi to log serial data...and grabserial seemed like it would make it easy.

Initially, I couldn't get grabserial to work properly...but figured out how to make it work.  I suspect there may be an issue, but I'll admit not being a programmer-type, it is possible that I'm just dumb and this is user error.  ;)

First...I obtained grabserial via wget from <https://github.com/tbird20d/grabserial/releases/latest> which is version 1.8.1.  

Second, I un-tar'ed it, then did the install as shown here <http://elinux.org/Grabserial>:

sudo python setup.py install

Executing grabserial as follows:

grabserial -d /dev/ttyAMA0 -b 19200 -e 30 -t -T -o test.txt

Returns:  

Error: serial device '/dev/ttyAMA0' does not exist

(Note:  being that this is a Raspberry Pi, I need to not that I've disabled the serial console...and was able to get other scripts to pull data from /dev/ttyAMA0 so I know it exists.)

Looking around on the internet, I saw another user who edited the source with his settings, then executed it with python.  So I did the same, downloaded the source and changed the following:

        sd.port="/dev/ttyAMA0"
        sd.baudrate=19200

Now when I execute it with python:

python grabserial.py -e 30 -t -T -o test.txt

Everything works perfectly.  I suspect there is something going on with how the /dev/ttyAMA0 is getting passed from the command line into the program?

thanks much,
ben, kd5byb

tbird20d, 2015-08-24:
>________________________________
> From: kd5byb <notifications@github.com>
>To: tbird20d/grabserial <grabserial@noreply.github.com> 
>Sent: Sunday, August 23, 2015 3:14 PM
>Subject: [grabserial] /dev/ttyAMA0 not found when using command line, but when edited in grabserial.py, works fine (#10)
> 
>
>
>Good evening!
>Was happy to find grabserial - I want to use a Raspberry Pi to log serial data...and grabserial seemed like it would make it easy.
>Initially, I couldn't get grabserial to work properly...but figured out how to make it work.  I suspect there may be an issue, but I'll admit not being a programmer-type, it is possible that I'm just dumb and this is user error.  ;)
>First...I obtained grabserial via wget from https://github.com/tbird20d/grabserial/releases/latest which is version 1.8.1. 
>Second, I un-tar'ed it, then did the install as shown here http://elinux.org/Grabserial:
>sudo python setup.py install
>Executing grabserial as follows:
>grabserial -d /dev/ttyAMA0 -b 19200 -e 30 -t -T -o test.txt
>Returns: 
>Error: serial device '/dev/ttyAMA0' does not exist
>(Note:  being that this is a Raspberry Pi, I need to not that I've disabled the serial console...and was able to get other scripts to pull data from /dev/ttyAMA0 so I know it exists.)
>Looking around on the internet, I saw another user who edited the source with his settings, then executed it with python.  So I did the same, downloaded the source and changed the following:
>sd.port="/dev/ttyAMA0" sd.baudrate=19200 
>Now when I execute it with python:
>python grabserial.py -e 30 -t -T -o test.txt
>Everything works perfectly.  I suspect there is something going on with how the /dev/ttyAMA0 is getting passed from the command line into the program?
>thanks much,
>ben, kd5byb

Thanks very much for the detailed bug report.  I haven't seen this issue, but it's been reported before and I haven't been able to reproduce it.  I suspect the problem is in serial.tools.list_ports.comports()

Can you please do the following on your system and send me the results:

At a shell prompt on your host system do:
$ python
>>> import serial
>>> print serial.tools.list_ports.comports()
^D

$ ls -l /dev/ttyA*

Thanks.
 -- Tim

kd5byb, 2015-08-24:
Hi Tim,

My pleasure:

pi@raspberrypi ~ $ python
Python 2.7.3 (default, Mar 18 2014, 05:13:23)
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> print serial.tools.list_ports.comports()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'tools'
>>>

pi@raspberrypi ~ $ ls -l /dev/ttyA*
crw-rw---T 1 root dialout 204, 64 Jan  1  1970 /dev/ttyAMA0
pi@raspberrypi ~ $

The above started me down the path of perhaps there is a serial.tools package that I don't have.  So I did some poking around...and found a post that suggested a slightly different method of listing comports:

pi@raspberrypi ~ $ python -m serial.tools.list_ports
no ports found

Very interesting!  Not sure what it all means...  ;)

thanks much,
ben

kd5byb, 2015-08-24:
Uh oh...I think I accidently just closed it...

kd5byb, 2015-08-24:
Got it back open.  Like I said...I'm sort of an idiot...  :)

tbird20d, 2016-05-10:
OK - I finally got around to providing a workaround for this.  I just added a -S option to allow skipping the serial tools device check.  This is kind of an ugly hack, but grabserial prints a message suggesting to try this option when it encounters the bug.  Hopefully this will guide people to work around the issue.

Sorry this took so long to close.  I realize you had a workaround, so it kind of dropped on my priority list.  I hope that grabserial is still providing a useful tool for you.

closed 2016-05-10 by tbird20d