tbird20d/grabserial #11: Cannot import grabserial as python module

andreMonkey, 2016-05-08:
Hello!
I ran the build and the install command for the setup.py, but cannot import grabserial as a module in a python script. I am new to python and a bit lost. Grabserial works amazingly from the terminal and is the only tool to read the values reliably, it would be super cool to be able to use it inside my python script.
Thanks a lot for any help! :)
Andre

andreMonkey, 2016-05-08:
import grabserial 
fails with the message "no such module named grabserial"
Here the response of the installer i@raspberrypi ~/grabserial $ sudo python setup.py install
running install
running bdist_egg
running egg_info
writing requirements to grabserial.egg-info/requires.txt
writing grabserial.egg-info/PKG-INFO
writing top-level names to grabserial.egg-info/top_level.txt
writing dependency_links to grabserial.egg-info/dependency_links.txt
reading manifest file 'grabserial.egg-info/SOURCES.txt'
writing manifest file 'grabserial.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
warning: install_lib: 'build/lib.linux-armv7l-2.7' does not exist -- no Python modules to install

creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/EGG-INFO
installing scripts to build/bdist.linux-armv7l/egg/EGG-INFO/scripts
running install_scripts
running build_scripts
creating build/bdist.linux-armv7l/egg/EGG-INFO/scripts
copying build/scripts-2.7/grabserial -> build/bdist.linux-armv7l/egg/EGG-INFO/scripts
changing mode of build/bdist.linux-armv7l/egg/EGG-INFO/scripts/grabserial to 755
copying grabserial.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying grabserial.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying grabserial.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying grabserial.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying grabserial.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/grabserial-1.8.1-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing grabserial-1.8.1-py2.7.egg
Removing /usr/local/lib/python2.7/dist-packages/grabserial-1.8.1-py2.7.egg
Copying grabserial-1.8.1-py2.7.egg to /usr/local/lib/python2.7/dist-packages
grabserial 1.8.1 is already the active version in easy-install.pth
Installing grabserial script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/grabserial-1.8.1-py2.7.egg
Processing dependencies for grabserial==1.8.1
Searching for pyserial==3.0.1
Best match: pyserial 3.0.1
Removing pyserial 2.6 from easy-install.pth file
Adding pyserial 3.0.1 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Finished processing dependencies for grabserial==1.8.1

tbird20d, 2016-05-10:
grabserial was really not written to be used as a python module, but rather as a standalone program.

However, having said that, it's relatively easy to convert it into a module.  I'm working on an update now that should allow this.  I hope to have it done by the end of the day.

andreMonkey, 2016-05-10:
Wow!
Thank you!!
That is so fantastic!

tbird20d, 2016-05-10:
OK - I've made some changes to support this.  Please see the latest version (1.9.1).   I haven't figured out what's going on with the setup.py script.  (This was contributed someone, and I've never understood the details very well.)  When you do 'sudo python setup.py install', it puts an 'egg' file in the site-packages directory, but it should be a .py file.

You can manually put this where it's needed just by copying grabserial and adding the .py extension.
I'm note sure where the site-packages are for the raspberry pi release, but here's what I did on Ubuntu 14.04:
sudo cp grabserial /usr/local/lib/python2.7/dist-packages/grabserial.py

Then from inside a python script (or the python intrpreter), I can do this:
  import grabserial
  grabserial.grab(["-d", "/dev/ttyACM1", "-o", "bootlog"])

It's klunky, but it appears to work.
If you can help me fix up the setup.py to correctly unpack the egg into a .py in the site-package directory, that'd be great.  But if not, that's fine.  I'll do some digging and see if I can figure out how to do this.

closed 2016-07-01 by tbird20d