tbird20d/grabserial #13: Misspelled variable name

DrASK, 2016-08-27:
https://github.com/tbird20d/grabserial/blob/master/grabserial#L199

`baudrate` -> `baud`

```
Traceback (most recent call last):
  File "grabserial", line 384, in <module>
    main()
  File "grabserial", line 178, in main
    print("Error: invalid baud rate '%d' specified" % baudrate)
NameError: global name 'baudrate' is not defined
```

DrASK, 2016-08-27:
I should also add, that it would be nice to have an option to skip the baudrate check as covered on the [pyserial documentation](http://pyserial.readthedocs.io/en/latest/pyserial_api.html):

> The parameter baudrate can be one of the standard values: 50, 75, 110, 134, 150, 200, 300, 600, 1200, 1800, 2400, 4800, 9600, 19200, 38400, 57600, 115200. These are well supported on all platforms.
> 
> Standard values above 115200, such as: 230400, 460800, 500000, 576000, 921600, 1000000, 1152000, 1500000, 2000000, 2500000, 3000000, 3500000, 4000000 also work on many platforms and devices.
> 
> Non-standard values are also supported on some platforms (GNU/Linux, MAC OSX >= Tiger, Windows). Though, even on these platforms some serial ports may reject non-standard values.

Apparently `sd.BAUDRATES` only contains the "standard" ones.

tbird20d, 2016-08-31:
Thanks for the bug report and the idea.  I added a new option -B to support setting the baudrate without the check.

See the the 1.6.3 version of the code (commit 7cbf104b) for the update.
 -- Tim

closed 2016-08-31 by tbird20d