tbird20d/grabserial #25: Specify file storage directory and format

obabarin, 2018-10-12:
hi tbird20d,

Nice application here, seems to be suitable for my application.

However, I am looking into specifying a directory for the grabbed files (using the app to log data from serial on a RPi3, and prefer to save the files to an external pen drive). I would expect that to be a parsed variable in -o "%". Do you have a plan of doing this in the future? Any current work around you may suggest?

obabarin, 2018-10-12:
hi,

I think I found a solution doing
grabserial -v -d /dev/ttyAMA0 -T -b 115200 -e 10 -Q -o "/home/pi/Documents/Grabber/file-%d-%H.%M.%S.csv" -a

tbird20d, 2019-02-27:
Sorry for the slow response.  I must have missed the e-mail notification for this.  It looks like grabserial can do what you want, so I'm closing this issue.  Regards.

closed 2019-02-27 by tbird20d