tbird20d/grabserial #26: Grabserial as startup service/file/bash script

obabarin, 2018-10-16:
I made a script to capture ttyAMA0 as follows..
`#! /bin/bash
sleep 5
grabserial -v -d /dev/ttyAMA0 -T -b 115200 -e 3600 -Q -o "/home/pi/file-%d-%m-%Y-%H.%M.%S.txt" -a
`

This runs well from the terminal but I can't get it to work as a startup script in either cron job or systemd service or even when I add it to update-rc.d

On update-rc.d, it just create the file, but does not read any data into it. As a systemd service, the script keeps looping- opens, doesn't store any data, closes, opens another files; all within seconds. I think this has to do with systemd configurations. In a cron job, the job just wouldn't start.

Is it possible grabserial was not written to be suitable as a startup?

obabarin, 2018-10-18:
hi tbird20d,

The issues is really from line 500 of the source code. But I wont be changing it as suggested. I added the bash script to rc.local and all seems to go on well for now. I will continue testing, and if any other issues comes up, I will try your suggestion or reach out to you again.

Thanks for the wonderful code. It does what I needed for the time being.

NB- apologies for the formatting of the text of the output and the bash script.

closed 2018-10-18 by obabarin

tbird20d, 2018-10-18:
No problem.  I'm glad you found a solution.

henning-schild, 2018-10-27:
@obabarin grabserial is great for debugging and measuring timing etc. if you are looking for a deamon to capture serial console output https://www.conserver.com/ might be the better solution ,will log to syslog do log rotation, probably comes with your distro including init/systemd stuff, and you can interact with the consoles ...