tbird20d/grabserial #30: AutoStart - Terminal - Grabserial - Append File

GT40Wins, 2019-03-11:
Hi Tim,  I'm new to Arduino, Raspberry & Python.  But that has not stopped me from taking temperature measurements w/ Arduino Mega and then using your **GREAT** program transferring them to a Raspberry Pi 3+ file.  Wanted to create a headless capability to do that. I'm learning Python fast but will NEVER in million light years will be as good as you are.  Can you help with the following:
1. Autostart (on power up) Grabserial with parameters [python grabserial -t >> <File Name>] but must be run in terminal so that must start first. I tried with lxterminal (see code below). Got grabserial to start but no way to pass parameters?  Looked at your code and most is greek (too bad it is not in fortran) Was thinking it could be configured to invoke parameters and file name w/o passing when first run via terminal? 
2. If this works the Data File would get written over each time the PI boots is there any way to invoke "a" append, or change file name for each session?
3. Can the file be located on a USB drive? Not sure how to pass drive location to grabserial. 
Sorry for being so needy but graduated from College with slide rule in hand.

**Best Regards & Thanks** 

```
[Desktop Entry]
Type=Application
Name=Headless.Desktop
Comments=Headless Commands
Exec=lxterminal -e python grabserial
Hidden=false
terminal=true     
```

tbird20d, 2019-03-11:
I'm not sure exactly what you are doing.  Is the Desktop Entry below a Gnome desktop file?
My understanding is that the 'Exec' line in the desktop file supports adding arguments to the command which is executed.
I haven't played around with it, but you might be able to invoke grabserial directly (not via lxtermina, and not via python).The grabserial script should have "#!/usr/bin/python" as the first line, so your Exec line might be able to be something like:Exec=grabserial <options>
You can put grabserial output to a file using the '-o' or --output argument.  If you specify a filename of '%' grabserialwill use a filename consisting of the current date and time (when it was started).
To put the output to a specific location in your filesystem, you can either 'cd' to that location before executinggrabserial, or specify the full path in the -o line.
I'm not sure where the USB drive would be located on your system.  Often, USB drivers are automatically mountedin a directory under /media/, and you would just need to use the path to that directory as the argument to -o.
It might be easiest to write a small shell wrapper to take care of this, and you can certainly customise thefilename that grabserial can use for output using a shell script and the 'date' command.
Like so: * create a shell script called: start_grabserial.sh * put the following lines in it:
#!/bin/shtodays_filename="/media/myusbdrive/temperature_log_$(date +%Y-%m-%d).log
grabserial <options> -o $todays_filename
 * put this in the Exec line in your desktop fileExec=/path/to/my/start_grabserial.sh

That should work, but I haven't tried it myself.
Looking at the code, it looks like if the filename contains % signs, then the strftime() functionis used to fill in the date information directly.  So you might get away with just puttingthis in the Exec line:
Exec=grabserial <options> -o /media/myusbdrive/temperature_log_%Y-%m-%d.log
And you can of course use other format characters for other date elements as well.See https://docs.python.org/2/library/datetime.html#strftime-strptime-behavior for descriptionsof the format characters.

Let me know if you have questions. -- Tim

 
 

   
   - Autostart (on power up) Grabserial with parameters [python grabserial -t >> ] but must be run in terminal so that must start first. I tried with lxterminal (see code below). Got grabserial to start but no way to pass parameters? Looked at your code and most is greek (too bad it is not in fortran) Was thinking it could be configured to invoke parameters and file name w/o passing when first run via terminal?

GT40Wins, 2019-03-12:
Not sure how to determine desktop used but did check Add/Remove Programs and found many Gnome packages installed as well as one called Debian Desktop (common files).   

While in terminal I tried "python grabserial --output=% -t -T" and it worked. Created file w/ date/time file name and running time/differential for each set of data. Takes care of one problem, that is when Pi starts it will not overwrite file since they all have unique date/time.  Progress !

Then tried the .sh wrapper and either I did not get the two line in correctly or Exec command would not start grabserial. The [Desktop Entry] was script I found on internet is located in /home/pi/.config/autostart.  The author of article said it could not start until desktop is running, then lxterminal could be opened and python grabserial could then be run.  But sadly w/o options.

 Went back and tried "Exec= lxterminal -e python grabserial --output=% -t -T" it ran but put output file in "-t".  Guess it does not like options.  

Any new thoughts on how to autostart grabserial with above options?  What about instructions as to how to edit and hard code --output=%, -t and -T into grabserial program.  I'd be happy to just autostart with new file name/options with program and data in /home/pi (that appears to be default location for both?) 

I appreciate the help, you know I need it and believe I'm very close to having everything working. Will be used to monitor new refrigerator temperatures, and have another device to show real and imaginary power used.  Might add the voltage, current and power usage into Arduino in the future. The raspberry Pi and your program is the bridge between the monitored data and Excel.  **Thanks Bunches**

tbird20d, 2019-03-12:
Try using two '%' signs in your Exec line, like so:Exec= lxterminal -e python grabserial --output=%% -t -T

 -- Tim

GT40Wins, 2019-03-13:
Tried % in Exec command and it just changed file name to "%" and then tried other changes to see if it could be done via the Desktop Exec command.  Tried as you suggested to create .sh file then added location/name to Exec command. you guessed it no luck.

So started to looking at grabserial code to find out how to hard code options (this is not fortran)
Was able to enable printing of time into file by doing the following:
      # changed from 0 to 1
         show_time = 1
         show_systime = 0

Next could not understand how to modify output file name to show date/time.  Thought I should modify this code, but nothing seemed to work:
         if opt in ["-o", "--output"]:
            out_filename = arg
            if out_filename == "%":
                out_filename = "%Y-%m-%dT%H:%M:%S"
            if "%" in out_filename:
                out_filename = datetime.datetime.now().strftime(out_filename)  

What a language, not easy to follow, Best Regards & Thanks

tbird20d, 2019-03-13:
I think you *are* getting arguments passed on the Exec line.  Otherwise, grabserialwouldn't be using the correct port.  that is, the '-d <serial-device> ' argument is working fine,if I've been following your descriptions of what's working so far for you.
You have to use  -o "%%" in the Exec line.  The Exec launcher should converts double percentto single percent.  That should cause '%' to be passed to grabserial,which in turn should result in out_filename being: '%Y-%m-%dT%H:%M:%S, which shouldthen get interpolated by datetime.datetime.now().strftime() into a filename with the correct filename.
If you tried it with double-percent, try it with single-percent.  Maybe the Exec launcheris not behaving as documented.  If you pass a double-percent to strftime() it willget converted to a single percent (and not any time-related values).
However, if you want to hardcode the filename to have the interpolated time format, just change the followingline near the top of the the grab() function
Remove this line:out_filename = None
and replace it with this line:out_filename = datetime.datetime.now().strftime("%Y-%m-%dT%H:%M:%S")

(and don't use the -o option on the command line (Exec line))
That will cause grabserial to always use a filename consisting of the time anddate.
 -- Tim

GT40Wins, 2019-03-14:
Hi Tim,
Opened grabserial in Thonny and ran fine w/o options(except -t is hard coded into program), here is what shell shows: (running time, delta time and 3-temp readings) (so did not appears to need -d to find serial device.
<<><><><><><><<><><><><><>
Python 3.5.3 (/usr/bin/python3)
>>> %cd /home/pi
>>> %Run grabserial.py
[0.000005 0.000005] <Arduino is ready>
[0.020792 0.020787] 66.5;66.70;65.913;
[10.022730 10.001938] 66.7;66.70;66.071;
[20.027897 10.005167] 66.7;66.70;66.071;
<><><><><><><><><><><><>

So I went on to try this via /home/pi/.config/headless.desktop and reboot. Failed to run.
<<><><><><><><><><><><><
[Desktop Entry]
Type=Application
Name=headless.desktop
Comment=Headless Commands
Exec=lxterminal -e pythyon3 /home/pi/grapserial -t >> wed_test
Hidden=false
terminal=true
<><><><><><><><><><><><><>

Tried using terminal to test command, found that this worked: running time and date/time file name
<><<><><><><><><><><><><><><><><><><><><>
pi@raspberrypi:~ $ python3 /home/pi/grabserial --output="%"
[0.000006 0.000006] <Arduino is ready>
[0.021630 0.021624] 67.6;67.65;66.858;
[10.022427 10.000797] 67.6;67.65;67.016;

File that was created !!
/home/pi/2019-03-13T17:21:32
><><><><><><><><><><><><><><>><<><><><>

So being very careful transfered the command to /home/pi/.config/headless.desktop file and then reboot: here is the headless.desktop file and as you guessed it worked!!!
<<><><><><><><><><><><><><><><><><><><><><><>
[Desktop Entry]
Type=Application
Name=headless.desktop
Comment=Headless Commands
Exec=lxterminal -e python3 /home/pi/grabserial --output="%"
Hidden=false
terminal=true

File that was created !!
/home/pi/2019-03-13T17:27:43
<><><><><><><><><><><><><><><><><><><><><>

So went on to try this Exec command:
<><><><><><><><><><><><><><><><><><><><><>
Exec=lxterminal -e python3 /home/pi/grabserial -t -T --output="%"

All Options worked (edited program and removed the -t hard code)

This file was created after a reboot:
/home/pi/2019-03-13T17:44:27
<><><><><><><><><><><><><><><><><><><><><>

Never thought it would be that easy (only possible with your help)
I'm a dangerous combination a novice that does not understand commands and syntax.  Good news is if anyone need to run grabserial headless all details are here.

Best Regards & Thanks!!!!

tbird20d, 2019-03-14:
I'm glad you were able to get it working, and that grabserial is useful for your project.  :-)
Regards, -- Tim

GT40Wins, 2019-03-15:
Hi Tim 
Let raspberry/arduino run overnight took a 300kB of temperature records. It works!!
Found that when I tried to copy records to USB drive it failed.
Short story is Fat32, best I can tell, does not like / ? < > \ * | "  ; +and most important : 

So I modified one line as follows:
            if out_filename == "%":
                out_filename = "%Y-%m-%dT%H-%M-%S"
            if "%" in out_filename:

Just added "-" between hour and min and min and sec.
T%H-%M-%S          

Best Regards & Thanks

closed 2019-09-04 by tbird20d