tbird20d/grabserial #31: Porting grabserial from Python2 to Python3

sinnykumari, 2019-08-23:
Python2 is going to retire very soon upstream on [January 1, 2020](https://pythonclock.org/) . As a result, with upcoming Fedora 31 and onward releases, Python2 packages are getting [removed](https://fedoraproject.org/wiki/Changes/F31_Mass_Python_2_Package_Removal) . Is there a plan to port grabserial from Python2 to Python3?

I tried to rebuild python-grabserial package as it is [with Python3](https://koji.fedoraproject.org/koji/taskinfo?taskID=37225944), but running it throws some error:
```
grabserial 
  File "/usr/bin/grabserial", line 190
    if opt in ["-d", "--device"]:
                                ^
TabError: inconsistent use of tabs and spaces in indentation
```

sinnykumari, 2019-09-03:
Friendly ping, any update?

tbird20d, 2019-09-03:
I would be happy to have grabserial work with Python3.  I already have accepted patches tofix some issues with Python3, and have done some additional work myself (particularly inthe area of unicode string handle) to have grabserial support Python2 and Python3.
Any patches you would like to send, to fix issues you encounter in Python3 supportwould be most appreciated.  Please note, however, that I will not accept any patchesthat remove Python2 support.  The tool must be backward compatible with thatinterpreter.
Regards, -- Tim

sinnykumari, 2019-09-04:
Thanks! I will try to put a patch when I get some time

tbird20d, 2019-09-04:
I re-ran my local tests, on both python2 and python3 and had no problems.Can I ask what version of grabserial you are using?
Please use the one from my github repository:https://github.com/tbird20d/grabserial.git

version 1.9.9, git commit 39144da
Let me know if you still see the error message with Python3.
Thanks, -- Tim

I just enhanced the selftest (test.sh), and refactored the code and added inlinedirectives to make grabserial.

sinnykumari, 2019-09-05:
> I re-ran my local tests, on both python2 and python3 and had no problems.Can I ask what version of grabserial you are using? Please use the one from my github [...]

Ah, I am using the source from v1.9.3  [released version](https://github.com/tbird20d/grabserial/releases) which is indeed very outdated. Didn't notice that we are tagging latest content without marking them as release. Using latest tag v1.9.9 (once available?, I see latest tag as 1.9.8) it may just build fine (will try tomorrow or over weekend)

tbird20d, 2019-09-06:
Oops.  I'm not very skilled with github.  I didn't realize anyone looked at the 'releases' tab.  I've gone aheadand made 1.9.9 a release.  Thanks for the reminder. -- Tim

sinnykumari, 2019-09-06:
Thanks for doing 1.9.9 release!
Latest update looks fine for both [Python2 Fedora 30](https://koji.fedoraproject.org/koji/buildinfo?buildID=1372994) and [Python3 Fedora 31](https://koji.fedoraproject.org/koji/buildinfo?buildID=1373016).

closed 2019-09-06 by sinnykumari