tbird20d/grabserial #32: Functionality to support continuous logging

AlphaSierraHotel, 2019-09-06:
Hi Tim.

I've been using grabserial for a couple of years now to do some continuous data logging from an Arduino-based sensor.  With the burgeoning IOT trend, I'm sure I'm not the only one doing that.  In support of that effort though, I tweaked the code to do a couple of things differently.

1. I added a command line option to append to the output file rather than overwrite it
2. When there is a date in the output filename, I close it and open a new one if the date has changed.

I've also configured a systemd setup that will start the process of serial data logging as soon as the port is available - all of which would help to address things akin to issue #30 

I am wondering if you would be open to considering a pull request or two to implement this functionality for the benefit of others like me that are interested in continuous data logging using grabserial.  The systemd setup could be in the form of either notes/documentation or an example file (or both).

Kind regards.
Allen

tbird20d, 2019-09-06:
Certainly.  Send some pull requests, or patches, my way.  I'll take a look and see if they make sense.
I'm OK with extending grabserial with useful features, as long as they don't detract from ease-of-use for the original intended use case.  It appears that a lot of people are using grabserial for data logging, ratherthan for bootup time monitoring (the original use case), and that's fine with me.
Maybe I should start a document describing different use cases in more detail.  I think your document could be the start of that.
Thanks. -- Tim

AlphaSierraHotel, 2019-09-06:
Two pull requests created for the code.  Give me a day or two to put some verbiage around the systemd config setup for better documentation.  I'll close this issue now too.

closed 2019-09-06 by AlphaSierraHotel