tbird20d/grabserial #36: Don't know how to fix this

helgard999, 2019-09-18:
Hello 
I would like to use the grabserial program but Im having trouble getting it to run properly.
I'm a newbie  on a Raspberry pie so most probably is all the issues my fault.

installed the repository as follows:

git clone https://github.com/tbird20d/grabserial.git
sudo python3 setup.py install
grabserial -d /dev/ttyS0 -T -e 20 -o "%"
so it is living in 
/home/pi/grabserial

and it is also in 
/usr/local/bin

looks like the installation is OK, and logs a file from the serial port. if I use the -a attribute it stops after the second log with the following output.
pi@raspberrypi:~ $ grabserial -S -d /dev/ttyS0 -T -e 5 --output="%Y-%m-%d_%H-%M-%S.log" -Q -a
Traceback (most recent call last):
  File "/usr/local/bin/grabserial", line 4, in <module>
    __import__('pkg_resources').run_script('grabserial==1.9.8', 'grabserial')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'grabserial==1.9.8' distribution was not found and is required by the application
pi@raspberrypi:~ $

I have no idea what could be wrong?
I see there is a V1.9.9 release, it doesnt look like that is the one that is installed?
It seems to want to use python2.7, python 3 is installed on Raspberry Pie

to update the version can re override what is installed with what is in the Source code tar.gz file?

tbird20d, 2019-09-19:
I'm sorry you're having problems getting it running.
Here are some random thoughts and ideas: - I don't ever use the setup.py program.  It was contributed by a user but I don't know if it works.  So I probably can't help debug that. - I usually just put the 'grabserial' script in ~/bin, or /usr/local/bin, and call it from the command line.     - you should be able to put it in any directory on the PATH - You say that RaspberryPI only has the python3 intepreter, but the error messages are from Python2.
If you go into the script and edit the first line, you can hard-code one interpreter used to execute the script.
For example, to force grabserial to be executed using python3, change the top line in the file to:#!/usr/bin/python3
Change that, and try running it again, and see what happens.
Good luck! -- Tim

tbird20d, 2019-10-24:
Having heard no feedback on my suggestions, I'm closing this issue.

closed 2019-10-24 by tbird20d

helgard999, 2019-10-25:
Hi Tim 
Sorry I followed your suggestions, but still wasn't able to get it going. Got busy with something and were not able to spend more time on it. Don't have other options at the moment will have to get into it again else but would still like to use your program.
Best regards
HR