tbird20d/grabserial #60: Shebang Python location could cause not find the interpreter

J-Rios, 2021-06-03:
## Context

Current shebang line in grabserial/putserial/setup.py files, that specify where to search by default for the Python interpreter, points to /usr/bin/python, but there is systems where the python binary could be in other path or could be multiple python versions, and can cause the scripts to not be executed directly:

```bash
./grabserial
-bash: ./grabserial: /usr/bin/python: bad interpreter: No such file or directory
```

## Change Recomendation

Modify current shebang:

```python
#!/usr/bin/python
```

To use **env**, that search in PATH enviroment variable fro the interpreter and is a best way for portability (find the interpreter in most systems):

```python
#!/usr/bin/env python3
```

What do you think about it?

Regards.

tbird20d, 2021-06-11:
I actually used "#!/usr/bin/env python2" a long time ago, and I had problems.  I don't recall what they were, and unfortunately,
the commit that changes that line (3bddb068) is silent on why the change back was made.

Given this history, I'm hesitant to change it.  I suppose we could try it and see what breaks.  I'd rather not switch to python3, though.  Although I test with both python2 and python3, I still do development on grabserial with python2 as the primary language.

J-Rios, 2021-06-12:
Maybe that problem just was related to thread creation order (maybe due closing exiting main thread before stein thread) and not to the shebang line. I agree to just try change it and see if there is any problem (I test it in current commit and didn't find any problem).

Regarding python2 vs python3... Due python2 is deprecated, it is recommended to move on into python3, however due you did a good work maintaining both version support, it could be ok to keep that way, supporting both, just we need to make sure to don't use/introduce some python2 specific code that could break python3 support or otherwise. But maybe we could use "python" in the shebang line, instead "python2" or "python3", and that can allow the user decide which version to use by just creating a Symlink for that:

```python
#!/usr/bin/env python
```

closed 2021-10-01 by github-actions[bot]