tbird20d/grabserial #63: The --nodelta option only works with --systime

keestux, 2021-12-28:
The subject says it all. The `--nodelta` does only work with `--systime`, not with `--time`.

```
$ grabserial -d /dev/ttyACM0 -v -t --nodelta
Opening serial port /dev/ttyACM0
115200:8N1:xonxoff=0:rtscts=0
Printing timing information for each line
Use Control-C to stop...
[0.000000 0.000000] main(): This is RIOT! (Version: 2022.01-devel-1387-g9e975)
[0.006011 0.006011] Hello World!
[0.007005 0.000994] You are running RIOT on a(n) nrf52840dk board.
[0.010996 0.003990] This board features a(n) nrf52 MCU.
^Cgrabserial stopped due to keyboard interrupt
```

```
$ grabserial -d /dev/ttyACM0 -v -T --nodelta
Opening serial port /dev/ttyACM0
115200:8N1:xonxoff=0:rtscts=0
Printing absolute timing information for each line
Use Control-C to stop...
[19:23:23.609021] main(): This is RIOT! (Version: 2022.01-devel-1387-g9e975)
[19:23:23.614825] Hello World!
[19:23:23.615820] You are running RIOT on a(n) nrf52840dk board.
[19:23:23.619809] This board features a(n) nrf52 MCU.
^Cgrabserial stopped due to keyboard interrupt
```

Is that on purpose?

tbird20d, 2021-12-28:
No.  This is not intentional.  I'll look into it.  I'm guessing it's an easy fix.

Thanks for the report!
 -- Tim

tbird20d, 2021-12-28:
I've fixed this in commit e1a05a6

closed 2021-12-28 by tbird20d