tbird20d/grabserial #69: Created file location

helgard999, 2022-09-28:
Hi
Im running grabserial from the rc.local file on raspberry pi on startup.

sudo grabserial -v -d "/dev/ttyUSB0" -b 500000 -w 8 -p N -s 1 -t -T -o "%" -a &
No problems, but the output is generated in the root directory which is not desirable.
I would like the file to be generated in  /home/pi/  

sudo grabserial -v -d "/dev/ttyUSB0" -b 500000 -w 8 -p N -s 1 -t -T -o /home/pi/"%" -a &
any change to the "%" just creates a file with % filename.

sudo /home/pi/grabserial -v -d "/dev/ttyUSB0" -b 500000 -w 8 -p N -s 1 -t -T -o "%" -a &
this tells me grabserial does not exist...

Please help, am I missing something?

tbird20d, 2022-09-28:
Thanks for the report.  I changed the code so that this should work now (that is, you should be able to use a path with a filename of '%' with the -o  option).  Please see  this commit: https://github.com/tbird20d/grabserial/commit/2808d5a21ec7fb880be3002585457b9bbed8c101

You should be able to replace the grabserial program on your system with the one from this master branch, and see the fixed behavior.

closed 2022-09-28 by tbird20d

helgard999, 2022-09-29:
Thanks Tim this is great service!

helgard999, 2022-10-14:
Hi
It looks like the issue has not been resolved completely. The filename is now correct.
I'm using a Pi with the grabserial command called at boot from the /etc/rc.local :
sudo grabserial -v -d "/dev/ttyUSB0" -b 500000 -w 8 -p N -s 1 -t -T -o /home/pi/"%" -a &
Upon boot it correctly created the new file in /home/pi/2022-10-13_16:31:31
but subsequent files (-o option) is created in the / directory.
/2022-10-14_00:00:02  the filename is correct though, but should be at /home/pi 

Another note:
On the pi the grabserial code is installed by apt-get in  /usr/bin

tbird20d, 2022-10-19:
Oops.  You're right.  I fixed the log path problem at the time of startup, but didn't add the same fix to the location where the logs are rotated.  Thanks for catching this and reporting it!  I just did another fix for this case. See commit f99dda1

helgard999, 2022-10-20:
Thanks Tim! Will test it