tbird20d/grabserial #70: Why there is no output?

sunhaotian0122, 2022-10-11:
```
`#dmesg | grep console
[    0.000000] ACPI: SPCR: console: uart,mmio,0x3f00002f8,115200
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.10.134-001.ali5000_1003_683de22baf_cbp.alios7.aarch64 root=LABEL=ROOT ro vconsole.font=latarcyrheb-sun16 vconsole.keymap=us biosdevname=0 console=tty0 scsi_mod.scan=sync nohz=off cgroup.memory=nokmem iommu.strict=0 virtio_ring.vring_force_dma_api=1 nokaslr crashkernel=0M-2G:0M,2G-256G:256M,256G-1024G:320M,1024G-:384M console=ttyS0,115200 ipv6.disable=1 rd.driver.pre=ahci LANG=en_US.UTF-8 softlockup_panic=1 numa_spinlock=off
[    0.000415] printk: console [tty0] enabled
[   10.203915] printk: console [ttyS0] enabled`
```

```
#grabserial -v -d /dev/ttyS0 -e 30 -t
Opening serial port /dev/ttyS0
115200:8N1:xonxoff=0:rtscts=0:rts=1:dtr=1
Program set to end in 30 seconds
Printing timing information for each line
Use Control-C to stop...
grabserial stopped due to time expiration
```

When I run grabserial, there is no output. I don't know if it's the cause of ttyS0? However, ttyUSB* does not exist on my computer. So, I use `dmesg | grep console` find ttyS0, can it replace ttyUSB0? Or, how can I find the corresponding serial port to replace the ttyUSB0?

tbird20d, 2022-10-11:
What type of machine is this dmesg from?  I'm guessing that it is x86-based since it's using ACPI.  It also appears that you are using a dedicated UART on the board (not from a USB device) as the console device.  Given that, I wouldn't expect you to have any ttyUSB* devices related to this console (on the machine where the dmesg is being run).

Your kernel command line has "console=/dev/tty0" (note this is NOT 'ttyS0' (with an 'S')).  I'm not sure if that matters or not.  You might look at /dev/console, and see what it's linked to, under /dev.

Also, just to clarify, is this dmesg from the device under test (the target), or the device watching the device under test (the host).  A computer can't use grabserial to watch it's own serial port.  The device on the target that is emitting the kernel console messages is different from the device on the host that is receiving the characters.  What is the actual hardware connection you have between the host and target?  This would help me diagnose the problem.

sunhaotian0122, 2022-10-12:
```
#uname -a
Linux j66e01291.sqa.eu95 5.10.134-001.ali5000_1003_683de22baf_cbp.alios7.aarch64 #1 SMP Mon Oct 3 20:20:26 CST 2022 aarch64 aarch64 aarch64 GNU/Linux
```
```
#grabserial -v -d /dev/tty0 -e 30 -t
Error: serial device '/dev/tty0' does not exist, aborting.
If you think this port really exists, then try using the -S option
to skip the serial device check. (put it before the -d argument)
```
```
#ls -l /dev
total 0
crw-r--r-- 1 root root     10, 235 Oct 10 11:57 autofs
drwxr-xr-x 2 root root         240 Oct 10 19:56 block
drwxr-xr-x 2 root root         140 Oct 10 19:56 bsg
drwxr-xr-x 3 root root          60 Jan  1  1970 bus
drwxr-xr-x 2 root root        2860 Oct 10 11:57 char
crw------- 1 root root      5,   1 Oct 10 11:57 console

#grabserial -v -d /dev/console -e 30 -t
Error: serial device '/dev/console' does not exist, aborting.
If you think this port really exists, then try using the -S option
to skip the serial device check. (put it before the -d argument)
```

In fact, this is an arm64-based machine. Because tty0 is not a serial device, so I did not mention it. It seems that the /dev/console is not linked to any file. This dmesg from the device under test (the target). 

I would like to know how I can find the corresponding serial port to use grabserial. In the example, there is only ttyUSB0 without telling how to find the corresponding serial port, if there is no ttyUSB0.

tbird20d, 2023-01-10:
can't reproduce

closed 2023-01-10 by tbird20d