tbird20d/grabserial #74: Date in filepath no longer works

pml-jpe, 2023-03-10:
I wanted to use subdirectories of year, month and day for output files so I used following option in grabserial (on Windows): `-o "C:\data\%%Y\%%m\%%d\%%FT%%H00%COM%.txt"`

I forked the repo and added code to make sure output directories would automatically be created if they didn't exist (otherwise grabserial quits with error, file not found).

I have just pulled latest version and I can see there has been substantial changes to the output filenaming. Now the output directories are created with literal strings `%Y`, `%m`, `%d` instead of actual year, month and date. I have had a look in the code around lines 525-536 but I can't quite understand where these are getting converted to strings. If you can point me to the correct code it would be a great help.

Thanks

pml-jpe, 2023-03-10:
Update: I think I have narrowed it down to the following lines 552-553:

`out_filename = datetime.datetime.now().strftime(out_pattern)
out_filepath = os.path.join(out_dirname, out_filename)`

So only the filename is having strftime applied, but the path does not. I will play around and see what I can do.

pml-jpe, 2023-03-10:
Final update: I have now fixed this with the following commit in my fork:

https://github.com/pml-jpe/grabserial/commit/0d7296a34a26c3cc8be609c37848023e891f2f75

Thanks

tbird20d, 2023-03-10:
Using time and date elements in the path to the output file was not something I anticipated.   I'll think about how to support this.  See my comment on your patch

tbird20d, 2024-07-09:
This was fixed in commit #75

closed 2024-07-09 by tbird20d

pml-jpe, 2024-07-10:
> This was fixed in commit #75

Thanks, I have pulled your changes into my fork and will test later.